nachocarballeda/gclass_amplifier #12: Zobel

anojo, 2015-12-14:
No va para el informe de aprobación, pero hay que ver después. Según Douglas, 10 ohms y 0.1uF va bien siempre y fuck it.

anojo, 2015-12-16:
La escribe "ZoEbel" en el mail

igpollitzer, 2015-12-16:
Agregué la red de Zobel. Dejo este inconveniente abierto, por si quieren dejarlo hasta que lo agreguen en el informe. Sino, cierrenlo.

anojo, 2015-12-16:
Claro que queda abierto. No le consultamos a Fede, no lo agregamos en el circuito.

anojo, 2015-12-16:
Además, por loq ue veo en el circuito del clase G del Douglas, además de esa RC tiene un RL paralelo a la salida

anojo, 2015-12-16:
Aparentemente, es el RC lo que se llama Zobel. Y el LR es para "aislar al amplificador de capacidad de carga". "Un componente de 15-20 vueltas de diámetro 1 pultada debería funcionar bien. El valor de la inductancia no es para nada crítico para la estabilidad"

anojo, 2015-12-16:
El inductor tiene que ser de aire para tener menos distorsión. Los valores de L suelen andar en 1-7uH.

closed 2015-12-22 by anojo