nachocarballeda/gclass_amplifier #34: Modelo como op-amp realimentado

anojo, 2015-12-16:
Analicen el circuito como si fuera un Op.Amp. realimentado, viendo las Zin, las Zo, etc, todo tal cual lo vieron en la 1era parte de la materia, incluído el SR, la compensación etc.

anojo, 2015-12-16:
Ok, no pusimos ningún diagrama con el triangulito realimentado. Pero slew rate, compensación, no estoy seguro de lo que quiere. ¿Ustedes?

igpollitzer, 2015-12-16:
asumo que quiere que hagamos el triangulito, con el cuadrado de realimentación abajo, midiendo tensión y sumando tensión (paralelo/serie) eso modifica las impedancias de entrada y salida. No se que es lo que quiere con el SR y la compensación, salvo que sea hacer Ivas x Cmiller.

anojo, 2015-12-16:
@igpollitzer pero el nodo de Miller está adentro del opamp, no veo cómo se puede mencionar con ese modelo

nachocarballeda, 2015-12-16:
[image]
Bob Cordell  - Designing Audio Power Amplifiers

igpollitzer, 2015-12-17:
Bueno, yo creo que esto ya está terminado. Hice el dibujito del opamp realimentado, hice el gráfico de Bode compensado, y el cálculo del Slew Rate. Todo Bertu-Style.

closed 2015-12-18 by anojo