nachocarballeda/gclass_amplifier #44: Modelos transistores 1era etapa

anojo, 2015-12-22:
Encontrar/inventar modelo de transistor para la 1era etapa. Si lo mejor que podés encontrar en un subcircuito, y no sabés cómo hacerlo funcionar en ltspice avisá, que para mi fue prueba y error

nachocarballeda, 2016-02-12:
Los transistores a utilizar en la etapa de entrada son 
MMPQ2907A (PNP)
MMPQ2222A (NPN)

Spice Model [http://hep.fi.infn.it/PAMELA/pdf/FFB2907A.pdf]
PNP (Is=650.6E-18 Xti=3 Eg=1.11 Vaf=115.7 Bf=231.7 Ne=1.829 Ise=54.81f Ikf=1.079 Xtb=1.5 Br=3.563 Nc=2
Isc=0 Ikr=0 Rc=.715 Cjc=14.76p Mjc=.5383 Vjc=.75 Fc=.5 Cje=19.82p Mje=.3357 Vje=.75 Tr=111.3n Tf=603.7p
Itf=.65 Vtf=5 Xtf=1.7 Rb=10)

Spice Model [http://www.farnell.com/datasheets/172298.pdf]
NPN (Is=14.34f Xti=3 Eg=1.11 Vaf=74.03 Bf=255.9 Ne=1.307 Ise=14.34f Ikf=.2847 Xtb=1.5 Br=6.092 Nc=2 Isc=0
Ikr=0 Rc=1 Cjc=7.306p Mjc=.3416 Vjc=.75 Fc=.5 Cje=22.01p Mje=.377 Vje=.75 Tr=46.91n Tf=411.1p Itf=.6
Vtf=1.7 Xtf=3 Rb=10)

nachocarballeda, 2016-03-15:
Done.

commit 5f070beb (2016-03-15): Cambiando modelos transistores entrada

Ahora los transistores de la simulacion son los que vamos a utilizar (integrados)
closes #44
closed 2016-03-15 by nachocarballeda